thelastWallE/OctoprintKlipperPlugin #4: Klipper configuration file is no more loaded in octoklipper settings.

lcha78, 2020-09-01:
Klipper configuration file is no more loaded in octoklipper settings.
It was working in 0.3.0

Since you rename path to configpath, I think that this line (26) is the problem in OctoprintKlipperPlugin/octoprint_klipper/templates/klipper_settings.jinja2 :
 "<input type="text" class="input-block-level" data-bind="value: settings.settings.plugins.klipper.configuration.path">       "
 should be
" <input type="text" class="input-block-level" data-bind="value: settings.settings.plugins.klipper.configuration.configpath">  "

AliceGrey, 2020-09-02:
@lcha78 Could you test the PR and let me know if that resolves the issue?

MoJo2600, 2020-09-02:
I can confirm that the configuration is shown in the PR. I have my configuration in ~/printer.cfg and I can see it on the settings page.

closed 2020-09-02 by AliceGrey

MoJo2600, 2020-09-02:
Buuut.. there seems to be another issue. If I save the configuration the files gets truncated and is then empty. I'm trying to figure out whats wrong.