thelastWallE/OctoprintKlipperPlugin #7: OctoprintKlipperPlugin update error

lcha78, 2020-09-02:
Good new, Octoprint detect the new version 0.3.1 of Octoklipper and propose to update. But there is a problem in the namming somewhere : OctoprintKlipperPlugin instead of OctoPrintKlipper

[image]

The whole error text 
Updating, please wait.
/usr/local/bin/python -m pip --disable-pip-version-check install https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip --no-cache-dir
Collecting https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip
ERROR: HTTP error 404 while getting https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip
ERROR: Could not install requirement https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip because of error 404 Client Error: Not Found for url: https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip
ERROR: Could not install requirement https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip for URL https://github.com/AliceGrey/OctoPrintKlipper/archive/0.3.1.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

AliceGrey, 2020-09-02:
@lcha78 Issue should be resolved by https://github.com/AliceGrey/OctoprintKlipperPlugin/commit/a2291b1a3d4796c5b8861f0efc20e4ed83a3de45

I have re-done the release tag for 0.3.1. Please try it now.

Thanks!

lcha78, 2020-09-03:
Should be OK. 
Have to wait for the next release to be sure.

frap129, 2020-09-15:
I was able to update to 0.3.1 without running into the issue mentioned above

AliceGrey, 2020-09-17:
Marking as resolved

closed 2020-09-17 by AliceGrey