thelastWallE/OctoprintKlipperPlugin #10: [Feature Request] Any chance of revealing some API commands?

NotExpectedYet, 2020-09-26:
Hi, thanks for taking up the development of this plugin! 

About your environment:
 - Distribution version: Any
 - Octoprint version: Any

Hi, I'm the developer of and Application called OctoFarm that combine mutliple OctoPrint instances into a single interface. I personally use this plugin and so do some of my users whome are asking me to support this plugin. I was wondering if it would be at all possible to enable API access to some of the klipper functions this plugin provides. 

Some of the possible features: 
1. Macro's, send a list of available macro's commands a user has setup. 
2. Klipper configurations viewing and editing. 
3. Grab Klippers current status

Absolutely no rush, I could possibly take a stab at this myself at somepoint.

AliceGrey, 2020-09-27:
@NotExpectedYet This definitely seems like a good set of features to have. Currently I'm prioritizing bugs that popup. I'd like to get the plugin more stable before I go adding more features. You or any other dev are welcome to create a PR for these features though.

NotExpectedYet, 2020-09-28:
@AliceGrey Brilliant, I will certainly have an attempt if I get time.

thelastWallE, 2021-05-04:
> @AliceGrey Brilliant, I will certainly have an attempt if I get time.

Anything came around with this?
If you give me a list like what data to provide on the api. I can look to get it in.
It will need definitely permissions for OctoPrint for some data.

NotExpectedYet, 2021-05-05:
@thelastWallE 
Saldy no. with going back full time and my own app I haven't had the time. 

Thank you for the offer! I will make a proper list and get back to you asap.

NotExpectedYet, 2021-05-09:
Hello! 

I've had a scout through the plugin a bit more to refresh my memory and I've listed below some possibilities and where they'd be nice to appear. Hope this helps. 

1. Klippers State: ie "Klipper: Standby" (Could be fed back through the websocket connection)
2. Host + Firmware restart actions (API)
3. List of available macros (API). With these been all gcode I don't think we will need a specific way to fire them as I can send them through the current functionality as I already do with the terminal. 
4. Tools Actions list, These could be just listed with the Macro's. (API)
5. Messages output. (Could utilise the websocket again for these similar to OctoPrints terminal)
6. Get the latest config, post back an edited version of the config. (API) I just checked some of the functions and it seems there might already be API calls for this one so might be doable without any modification. I often forget to look at the network tab in OctoPrint when I click stuff. 

I've ordered them in somewhat of a "required" > "nice to have" way as I've no idea how possible these would be currently. Honestly any of those would be a great start so thanks for the offer. 

Also nice work on the new updates! Love how polished the plugin is looking now.

closed 2021-05-19 by thelastWallE