thelastWallE/OctoprintKlipperPlugin #15: Unclear on the forking - which is the correct one?

marcosscriven, 2020-10-15:
Just saw this message in Octoprint:

```Version 0.3.1 of this plugin is available from a new maintainer, but your version still looks for updates at the repository of the old maintainer.```

However, the original repo https://github.com/mmone/OctoprintKlipperPlugin by @mmone doesn't mention this fork.

How is the selection of forks managed for a single plugin? Who chooses which is in the Octoprint plugin repo?

AliceGrey, 2020-10-21:
This is the fork that is now listed on the octoprint website(https://plugins.octoprint.org/plugins/klipper/). The original developer mmone has abandoned the project.

closed 2020-10-21 by AliceGrey

marcosscriven, 2020-10-21:
Is there some governance around that? Also, I see the licence was changed - I didn't think it was possible to strip GPL from the original source?