thelastWallE/OctoprintKlipperPlugin #16: Konfiguration editor does not reload after SAVE_CONFIG command

ohumi, 2020-10-19:
Before submitting your issue please make sure to provide the following information 

About your environment:
 - Distribution version. octopi
 - Octoprint      OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0

- An OctoPrint log file. You can access and download it in the OctoPrint settings under the "Logging" menu entry.

Thanks

AliceGrey, 2020-10-21:
@ohumi Could you provide more details of the error? Steps to re-create and a log file?

TheNitek, 2020-10-30:
I got the same problem:
I do a DELTA_CALIBRATE and SAVE_CONFIG afterwards. Locking at the config using SSH I can see the new config values. Looking at the config using OctoKlippers web editor, I still see the old configuration.

gufi, 2020-11-24:
a temporary workaround is to ensure that you hard refresh the entire octoprint ui. then go back in and new settings that were generated will be there. This is likely due to some kind of browser caching or something

ururk, 2021-02-08:
Just came here to report this bug and can confirm the behavior. I was using the PROBE_CALIBRATE routine, and after doing SAVE_CONFIG, went to this plugin's page and my new configuration was not visible - refreshing the page and the configuration appeared.

The first time this happened I lost my probe settings, as I had ran PROBE_CALIBRATE, updated a config setting, all without reloading the page. If you don't reload the page, it will overwrite your existing config even if it has changed on disk while OctoPrint was running.

ururk, 2021-02-08:
It feels like it should read the config anytime the plugin is invoked (not sure what the programming flow is for OctoPrint plugins, and if you have to deal with caching on your end).

gufi, 2021-02-08:
from what i can tell configs are loaded once since all plugins have their state held on the browser until refresh. There probably needs to be a little work between both Octo and plugin creators to allow a different behavior for plugins that can modify their settings externally without a browser

jangrewe, 2021-02-09:
I wouldn't call it a bug, but more of a feature request: either reload the config from disk when accessing that tab, or provide a button to manually reload the config - the later probably being the safe choice, so that users don't loose their manually typed in configuration just because they're switching tabs.

But yes, please add some kind of functionality to reload the config without having to reload the whole page! 🙂

thelastWallE, 2021-03-25:
Just an update on this. I have it almost ready, but i wanted to implement this along with the ace editor.
The editor would reload the file from disk as the settings dialog is opened.

closed 2021-03-26 by thelastWallE