thelastWallE/OctoprintKlipperPlugin #17: Klipper config extends beyond the Octoprint configuration window

dewi-ny-je, 2020-11-09:
Octoprint 1.4.2
Octoklipper 0.3.2

Windows 10, resolution 1600x900, zoom 125%.

The text box containing the Klipper config file extends below the Octoprint window, making the edit quite cumbersome especially on laptop, where scrolling with the touchpad is more involved.

See in the screenshot the double scrollbar.
[image]

The same happens with the Plugin list of Octoprint so I'm not sure you can solve it, but if possible, it would be appreciated.

AliceGrey, 2021-01-11:
@dewi-ny-je That's simply how the web UI works. For editing the config on a smaller monitor I might suggest copying the config file to another editor you can full screen. The configuration editor in the plugin is meant for small edits, checking what you have things set to, etc.

closed 2021-01-11 by AliceGrey

dewi-ny-je, 2021-01-11:
@AliceGrey but Octoprint had the same issue with the plugins section of the preferences (double scrollbar) and I think they solved it with the latest version. 
I'm no web developer, but isn't Octoprint offering a variable indicating the current height of the popup preferences window you can use to adapt your text field?