thelastWallE/OctoprintKlipperPlugin #18: Klipper virtual SD card integration

ssorgatem, 2020-11-10:
This is a feature request. Kinda. 
Not even sure if it's possible :P

Klipper supports a Virtual SD card that is typically set to be the Octoprint uploads directory.
However, when printing "from SD card" in Octoprint, there are several features that don't work, like print time estimation.

But if OctoPrint knows about Klipper, wouldn't it make sense that it recognized the "SD card" file and the local file with the same name as the same file? So even if starting the print from the "SD card", it could pull GCODE analysis and time estimation from the local file. Or when printing from the local file, having the option to actually print from the SD card file with the same name, whichever makes more sense or is more feasible.

This is just an idea I just had.

Would this feature be welcome in this plugin? Since it's klipper-related (I don't think this would be useful outside of klipper), I think it belongs here.

I'm probably going to start working on this myself, though I will also probably have no idea what I'm doing :P
This issue is mainly so that I can decide between starting an independent plugin or forking this to later do a pull request ;)

grigi, 2020-12-01:
I was thinking of the same requirement. When I print, I always want the SD version to print (as it has much less risk of pauses), but would love the local gcode analysis to happpen as well.

This, without having to force-refresh the SD, find it, double-check that it's what I wanted to print. That would be grand.

AliceGrey, 2021-01-06:
@ssorgatem I think this might be somewhat out of scope of  what this plugin currently does. This would more likely be an addition/change to octoprint/klipper itself. But if anyone comes up with a solution PRs are always welcome.

closed 2021-05-13 by github-actions[bot]