thelastWallE/OctoprintKlipperPlugin #19: Feature request: Button or shortcut to editing Klipper configuration file

Kollimolli, 2020-11-24:
Hi!

Currently to edit klipper configuration file (printer.cfg) i need to first open Octoprint settings which opens slowly, then Octoklipper settings and after that select Klipper configuration. Sometimes it’s frustrating if you have to edit it many times in a short time, ie. when testing or doing calbrations etc.

Is it possible to add button or shortcut to main page or to Klipper tab that open printer.cfg with a single click?

thelastWallE, 2021-01-11:
I just saw that Octolapse has something like that. A shortcut to the settings-dialog of the plugin. This would be sufficient?
I will look into it. Seems like octolapse has a js function for this button.
[image]

thelastWallE, 2021-01-12:
update: i have it almost ready. Just some testing with the new access settings of octoprint.
i took for now the same icon as octolapse but i think i write it just as a link like "printer.cfg".

PR #29 (2021-03-11): Shortcut to printer.cfg with granulary access
closes #19 

This displays a link, if someone is allowed to config klipper, on the klipper tab for the printer.cfg,
It opens directly the plugin tab for OctoKlipper with the tab "Klipper Configuration".
I tested it with a new user i made with only "Read-only Access". This user wouldn't then even see the link for the printer.cfg on the tab.

We can also add this function for all the other things that are displayed now but shouldn't if someone has no access for them. Like the list of macros for example.
[image]
closed 2021-03-11 by AliceGrey