thelastWallE/OctoprintKlipperPlugin #20: Not compatible with Octoprint 1.5.0?

louisferreira, 2020-12-02:
Just upgraded Octoprint to 1.5.0, then installed OctoKlipper 0.3.2 from the plugin manager, and it seems it is not compatible. Checking the Octoprint logs, I find the error message:
...
**2020-12-02 06:53:14,206 - octoprint.plugins.klipper - ERROR - Error: Klipper config file not found at: /home/pi/printer.cfg**
2020-12-02 06:53:14,383 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octopi2/'}
**2020-12-02 06:53:15,793 - octoprint.plugins.klipper - ERROR - Error: Klipper config file not found at: /home/pi/printer.cfg**
2020-12-02 06:53:15,981 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Ver 1.70.1 BL (Creality3D)"
...

Can this be confirmed?

kamahat, 2020-12-06:
I'm using Octoprint 1.5.1 and octo klipper 0.3.2 and no issue

mattbruman, 2020-12-08:
Same issue, cant update

`ERROR: HTTP error 404 while getting https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip
ERROR: Could not install requirement https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/mmone/OctoprintKlipperPlugin/zip/0.3.2
ERROR: Could not install requirement https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/mmone/OctoprintKlipperPlugin/zip/0.3.2 for URL https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip`

Master92, 2020-12-09:
> 
> 
> I'm using Octoprint 1.5.1 and octo klipper 0.3.2 and no issue

Same here, OctoPrint 1.5.1, OctoKlipper 0.3.2 and klipper 0.9.1 and no issues so far.

AliceGrey, 2021-01-06:
@louisferreira The error says you're missing printer.cfg in the /home/pi/ directory. Did you setup Klipper properly?

AliceGrey, 2021-01-11:
Closing issue as not related to the plugin but the usage of Klipper itself.

closed 2021-01-11 by AliceGrey