thelastWallE/OctoprintKlipperPlugin #26: Update from 0.2.5 to 0.3.2 Fails on OctoPrint 1.5.2

robschwieb, 2020-12-11:
This is what I receive when attempting to update:

`Updating, please wait.
+++++++++++++++++++++++++++++++++
Now updating OctoKlipper to 0.3.2
+++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip
HTTP error 404 while getting https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip
Could not install requirement https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/mmone/OctoprintKlipperPlugin/zip/0.3.2
Could not install requirement https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/mmone/OctoprintKlipperPlugin/zip/0.3.2 for URL https://github.com/mmone/OctoPrintKlipper/archive/0.3.2.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.`

**Here is the snippet from the octoprint.log**
`2020-12-11 23:21:15,018 - octoprint.plugins.softwareupdate - INFO - Starting update of klipper to 0.3.2...
2020-12-11 23:21:22,144 - octoprint.plugins.softwareupdate - ERROR - Update of klipper can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1788, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 144, in perform_update
    "Error while executing pip install", (stdout, stderr)
UpdateError: Error while executing pip install
2020-12-11 23:21:22,764 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': '0.3.2', 'from_version': '0.2.5', 'target': u'klipper'}
`

AliceGrey, 2021-01-06:
@robschwieb That's a pretty old version and it looks like you're still running 2.7. Please try a fresh install of the plugin.

collse, 2021-01-07:
> @robschwieb That's a pretty old version and it looks like you're still running 2.7. Please try a fresh install of the plugin.

thats pretty much did the trick for me as well ...

closed 2021-01-11 by AliceGrey