thelastWallE/OctoprintKlipperPlugin #34: Not working with Octopi 0.18.0

olivierD-FR, 2021-02-22:
Octopi 0.18.0 / Octoprint 1.5.3 / Python 3.73

Hello- I can install the plug in and it reads the status from the printer but cannot add macros nor vizualise the config. It used to work perfectly with previous Octopi image. Any idea ?

Thanks !

thelastWallE, 2021-03-06:
I/We need more information here.

You had me to go using my only sdcard, for now, to flash OctoPi 0.18.0 on it and try it myself.
I installed it. Updated to 1.5.3 OctoPrint. Got my backup in. And installed klipper. With this https://www.klipper3d.org/Installation.html And all was fine. Even before installing klipper the macros where all fine
Before i had copied the printer.cfg on /home/pi/printer.cfg, there was shown an "empty" textfield on the OctoKlipper config tab.
The command was this: `cp ~/klipper/config/example-cartesian.cfg ~/printer.cfg` You need to make sure you are in the home/pi/ directory atm.

thelastWallE, 2021-03-06:
I have copied my backuped printer.cfg over the copied default one after this.
Also you need to restart the system to get octoklipper to look for the new file(cache thing from the browser).

endurance1968, 2021-04-23:
I am using V18 as well - works.

closed 2021-05-04 by thelastWallE