thelastWallE/OctoprintKlipperPlugin #35: Klipper sidepane won't expand

adman953, 2021-03-01:
Before submitting your issue please make sure to provide the following information 

About your environment:

[octoprint (1).log](https://github.com/AliceGrey/OctoprintKlipperPlugin/files/6064511/octoprint.1.log)

 - Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

the octoKlipper side pane shows up but won't expand. i have tried disabling all of my addons and it still won't expand, it's very handy for the macros i have set up. thank you

thelastWallE, 2021-03-05:
In your logfile is an error message about plugin filamentmanager every time octoprints boots. Can you look if it is really disabled?

adman953, 2021-03-05:
It is actually installed, I reinstalled it and uninstalled it with clean up
data but that did nothing

thelastWallE, 2021-03-06:
Can you provided an logfile?
Have  also the option debug on. And it woud help to enable the extra klipper logmessages.
[image]

adman953, 2021-03-06:
> Can you provided a logfile?

here is the new log
 octoprint.log
<https://drive.google.com/file/d/1x-gUbZyJ8s50mRHojd7cKlmYntIfLEnz/view?usp=drive_web>

adman953, 2021-03-08:
thanks for all of the help but i figured out it was octoscreen. some sort of conflict there but i cant find exactly what.

thelastWallE, 2021-03-08:
i will look into that if i have time.

thelastWallE, 2021-03-08:
They are actively hiding the connection panel of octoklipper. https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/blob/32f6672beefa4fd28055759bd73b6493a7b369d1/octoprint_zbolt_octoscreen/static/js/zbolt.js#L41
```
 ZBoltOctoScreen.ConnectionViewModel = function (parameters) {
        var self = this;

        self.onAfterBinding = function () {
            var connection = $("#sidebar_plugin_klipper");
            connection.collapse("hide");
        }
    }
```
I don't know why they are doing this, but I did find the problem. The accordion menu is not expanding if the site was loaded with the menu collapsed. If i change the height of the `<div id="sidebar_plugin_klipper" class="accordion-body collapse" style="height: 0px;">` to over 0px. The menu is expanding again. This seems maybe to be also a problem with the whole octoprint accordion menu.

thelastWallE, 2021-03-09:
I changed the eventfiring a bit back for there hiding of the sidebarmenu. It is now hiding the menu on the first siteloading and you can expand it afterwards again. If you need it somehow other i would recommend to talk to the https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen dev-team.
My changes are here: https://github.com/thelastWallE/OctoPrint-Z-Bolt-OctoScreen/tree/klippersidebar

thelastWallE, 2021-03-09:
@adman953 Changes got merged into original OctoPrint-Z-Bolt-OctoScreen. You can install it like you did before over URL: `https://github.com/Z-Bolt/OctoPrint-Z-Bolt-OctoScreen/archive/master.zip`

closed 2021-03-14 by thelastWallE