thelastWallE/OctoprintKlipperPlugin #39: 0.3.2 to 0.3.3 update fails in Octoprint

ProfEngr, 2021-03-27:
The updater returns a URL error when trying to update to the newest (0.3.3) version. 

`2021-03-27 13:10:22,027   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip --no-cache-dir
2021-03-27 13:10:25,095 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-03-27 13:10:25,096 > Collecting https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip
2021-03-27 13:10:25,100 ! ERROR: HTTP error 404 while getting https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip
2021-03-27 13:10:25,101 ! ERROR: Could not install requirement https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/ShohninDmitriy/OctoprintKlipperPlugin/zip/0.3.3
2021-03-27 13:10:25,103 ! ERROR: Could not install requirement https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/ShohninDmitriy/OctoprintKlipperPlugin/zip/0.3.3 for URL https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/archive/0.3.3.zip
`

thelastWallE, 2021-03-28:
Did you updated that over the update plugin in octoprint?

The URL should point to this repository.

ProfEngr, 2021-03-28:
I just clicked the update button on the notification that popped over the UI.

thelastWallE, 2021-03-28:
Try to force the update.
[image]

thelastWallE, 2021-03-30:
Seems like this fork you got the 0.3.2 version from entered the url from the fork into setup.py.
[See here](https://github.com/ShohninDmitriy/OctoprintKlipperPlugin/blob/2041cdc64f71ec5ab39a0c9cb060294c1089bf54/setup.py#L30)  

You should reinstall OctoKlipper with the URL or zip upload Option to make sure you are on the official fork again.
"https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/tags/0.3.3.zip"  

[image]

closed 2021-03-30 by thelastWallE

ProfEngr, 2021-03-30:
OK. Will try that. I think I was on a fork due to an issue w/ the cfg getting corrupt from the UI. I use winSCP for editing so it doesn't matter now.