thelastWallE/OctoprintKlipperPlugin #41: Config not saving on last version

rstanislav, 2021-04-02:
Octoprint 1.5.3
Octoklipper last version
After update config files is not saving - it just resets to prev values and nothing saved.

`2021-04-02 18:27:22,552 - octoprint.server.util.sockjs - INFO - User rstanislav logged in on the socket from client ::ffff:192.168.1.33
2021-04-02 18:27:22,748 - tornado.access - WARNING - 404 GET /plugin/octolapse/getSnapshot?file_type=thumbnail&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1617384443948 (::ffff:192.168.1.33) 1.78ms
2021-04-02 18:27:22,753 - tornado.access - WARNING - 404 GET /plugin/octolapse/getSnapshot?file_type=snapshot&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1617384443948 (::ffff:192.168.1.33) 1.46ms
2021-04-02 18:29:24,654 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:29:24,661 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 269.28ms
2021-04-02 18:29:40,144 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:29:40,152 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 23.54ms
2021-04-02 18:30:35,275 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:30:35,282 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 17.57ms
2021-04-02 18:30:47,133 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:30:47,141 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 14.92ms
2021-04-02 18:31:12,941 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-04-02 18:32:33,128 - octoprint.server.util.flask - INFO - Passively logging in user rstanislav from ::ffff:192.168.1.33
2021-04-02 18:32:33,130 - octoprint.access.users - INFO - Logged in user: rstanislav
2021-04-02 18:32:33,736 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.33
2021-04-02 18:32:34,527 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.33
2021-04-02 18:32:34,544 - octoprint.server.util.flask - INFO - Passively logging in user rstanislav from ::ffff:192.168.1.33
2021-04-02 18:32:34,545 - octoprint.access.users - INFO - Logged in user: rstanislav
2021-04-02 18:32:36,887 - octoprint.server.util.sockjs - INFO - User rstanislav logged in on the socket from client ::ffff:192.168.1.33
2021-04-02 18:32:37,328 - tornado.access - WARNING - 404 GET /plugin/octolapse/getSnapshot?file_type=thumbnail&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1617384758532 (::ffff:192.168.1.33) 1.63ms
2021-04-02 18:32:37,334 - tornado.access - WARNING - 404 GET /plugin/octolapse/getSnapshot?file_type=snapshot&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1617384758533 (::ffff:192.168.1.33) 1.55ms
2021-04-02 18:32:38,417 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
[... 18 lines cut ...]
2021-04-02 18:32:46,931 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 16.53ms
2021-04-02 18:32:59,599 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:32:59,606 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 14.08ms
2021-04-02 18:33:22,262 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin klipper
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 428, in on_api_command
    if not self.validate_configfile(data["config"]):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_klipper/__init__.py", line 533, in validate_configfile
    dataToValidated.read_string(dataToBeValidated)
AttributeError: RawConfigParser instance has no attribute 'read_string'
2021-04-02 18:33:22,271 - tornado.access - ERROR - 500 POST /api/plugin/klipper (::ffff:192.168.1.33) 20.73ms`

Lefuneste83, 2021-04-02:
Same issue here ! Totally annoying !

thelastWallE, 2021-04-02:
Should be fixed in 0.3.7

closed 2021-04-02 by thelastWallE

rstanislav, 2021-04-04:
Broken again in 3.7.1

DefiantGlobe, 2021-07-09:
Broken still on 0.3.8.3