thelastWallE/OctoprintKlipperPlugin #42: Config editor does not allow duplicate blocks.

mtw3d, 2021-04-04:
Hi there! I love the new changes to the config editor, it is a huge improvement! 

That said, there is one annoying bug. 

Klipper allows you to create multiple blocks for the same settings. One handy usage for this is to let you have all your commonly edited settings at the beginning of the config file, so you don't need to wade through the entire file to find common settings.For example:

```
[bltouch]
z_offset: 1  

[additional setup]

[bltouch]
sensor_pin: ^BLTSEN
control_pin: BLTCTL
pin_move_time: 0.680
y_offset: -27
x_offset: -35
```

Unfortunately, your new editor now throws an error on this block and does not allow the usage. Can you either disable this check, or rather than just not allowing it, prompt so the user can decide whether it is intentional or not?

Also, I noticed a purely aesthetic bug. Klipper allows comments at the end of a line, such as 

`[X_stepper] # X Stepper Motor`
 
In the editor, if you have a line with a inline comment like that, the lines after it show up in pink, as if they were commented out. I assume the function properly, so this is not a huge problem, but I figured I would raise it!

thelastWallE, 2021-04-04:
Love those edge cases. Never knew that you can do this.
I will put an exception for this type of error.

thelastWallE, 2021-04-04:
fixed in https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.7.1
Take the option "Advanced Options" > "Force check for update" on the "Software Update" tab in OctoPrint to get the update immediately.

closed 2021-04-04 by thelastWallE

mtw3d, 2021-04-04:
Thank you for the quick fix!

thelastWallE, 2021-04-04:
Unfortunately i must continue working on this. Every parsererror after a duplicateconfigsection will not be excepted.

mtw3d, 2021-04-04:
It seems to be working for me now, but I appreciate your diligence!

thelastWallE, 2021-04-09:
I fixed the comments on section lines
e.g. [stepper x] # This is my stepper x
You need to switch the release channel to "Release Candidates" on the Update Plugin of OctoPrint to test it.
I also switched the parser to not use the strict mode. this helps for double sections and double options which are allowed in klipper.

mtw3d, 2021-04-09:
Thank you for your quick response!