thelastWallE/OctoprintKlipperPlugin #43: Broken again in 3.7.1

rstanislav, 2021-04-04:
Broken again in 3.7.1

_Originally posted by @rstanislav in https://github.com/thelastWallE/OctoprintKlipperPlugin/issues/41#issuecomment-813065448_

SelfDestructor, 2021-04-04:
Can confirm, unable to save the config file after updating to 3.7.1.

ev0kblitz, 2021-04-04:
yup i can't save  config after updating 3.7.1

SelfDestructor, 2021-04-04:
Looks like @thelastWallE got a fix up already! Will try it once my printer is done printing.

https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.7.2

3v0k, 2021-04-04:
thanks @SelfDestructor ^^
 it's works now (release candidate for update)

thelastWallE, 2021-04-05:
Had changed too much for the fix of duplicate sections. And the parsing check was only true if you had duplicate sections in the file. Sry for that.

fixed in https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.7.2

closed 2021-04-05 by thelastWallE