thelastWallE/OctoprintKlipperPlugin #45: [FR] Klipper "language" coloring

Sineos, 2021-04-06:
Thanks for keeping up the work on this plugin. Your additions and improvements are really appreciated.

I really do like the new editor with its language coloring scheme. If possible, I have two requests:
* The lines below the `#*# <---------------------- SAVE_CONFIG ---------------------->` are treated a comment. Would be cool to have the same coloring applied
* My eyes are getting older (depressing insight, I know 😏 ) so I would love to have the font size a tad bigger. Somewhere between current and Braille would be cool.

Thanks for considering.

thelastWallE, 2021-04-07:
I work on a "zooming" function there you can increase and decrease the font-size.
As for the Save_Config Section this would be lower priority.
Problem is that these lines are comments. These lines unfortunately start all with #*# .  And i would have to do some work to distinguish between comments and real options that klipper are reading.

Sineos, 2021-04-07:
Thanks for your comments. Any improvement is appreciated. My, maybe naive, idea was to distinguish between a single `#` as comment and `#*#` as Klipper `SAVE_CONFIG` in the parsing.

thelastWallE, 2021-04-11:
You can test a first zooming function on the release candidate channel.
[How to use the release channels to help test release candidates](https://community.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402#heading--plugins)
https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.8rc2

Sineos, 2021-04-12:
Works nicely on my side. Setting is saved as well and survives Octoprint restarts. Many thanks.

closed 2021-04-12 by Sineos

thelastWallE, 2021-04-12:
SAVE_CONFIG section should now have a syntax highlight with 0.3.8rc3

Sineos, 2021-04-14:
Works. Many thanks. 👍 

Just one small cosmetic thing:
[image]

TBH, not worth any additional efforts