thelastWallE/OctoprintKlipperPlugin #46: [Feature Request] File explorer for custom .cfg file

NovaViper, 2021-04-07:
Hi! I've just come from Fluidd and I'm wondering if OctoKlipper could add in support for the ability to edit klipper configuration files that are not under `printer.cfg. In my case, I have a `printer.cfg` file that loads in subfiles like `machine.cfg` and `macro.cfg` where `machine.cfg` has all of my machine definitions and `macro.cfg` has all of my macro definitions. In Fluidd, it shows a file explorer that shows the contents of my klipper_config folder like the following screenshot

[image]

thelastWallE, 2021-04-09:
I will work on that. But i need a little bit of researching of the work with ace editor. But i have a plan generally already in my head.

NovaViper, 2021-05-20:
> I will work on that. But i need a little bit of researching of the work with ace editor. But i have a plan generally already in my head.

Any progress on it? I don't mind giving any experimental stuff a try!

thelastWallE, 2021-05-21:
Unfortunately i don't have much time atm. Wife is sick and i need to care for our two children and two dogs. I started on something but it is not useable at all.

NovaViper, 2021-05-21:
Oh no! I hope your family gets better! ❤️

closed 2021-06-27 by github-actions[bot]