thelastWallE/OctoprintKlipperPlugin #48: Syntax Highlighting

Sineos, 2021-04-16:
Version: 0.3.8rc4
        
The Syntax Highlighting is pretty cool and thanks for your mission to get it perfect 👍 

Some minor weirdness I noticed:

[image]

[image]

[image]

Don't get me wrong: Feel free to simply close this issue and tell me to get lost with this crap 😏

thelastWallE, 2021-04-16:
No all fine. Will be even better as i get more configurations from users.

I will look into it. Didn't see anything on my configuration. Seems like some values are missing on the highlighter that are allowed to put as values.

jalabaya, 2021-05-04:
I realise this is closed, but I found this mismatched colour for AD597 in the latest version. Unsure if that is fixed by the commit ... Thanks!

[image]

thelastWallE, 2021-05-04:
you can test it with https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/rc.zip
this will have the latest commit on the rc branch in it.
I‘m not at the pc atm.

jalabaya, 2021-05-05:
Thanks for that! According to the plugin manager I now run OctoKlipper (0.3.8rc8) after a restart of the server. Nice additions! The single purple letter is still there though. I am not a programmer, but let me know if you need more information (logs etc.)

(Not related to this issue, but rc8: It would be nice to be able to set a "default" zoom level for the config viewer. It now resets, for example after looking at the general settings.)

thelastWallE, 2021-05-05:
try https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/heads/rc.zip

The zoom level should be saved if you click on the "Save" button on the OctoPrint settings dialog.

jalabaya, 2021-05-05:
This fixed it. Thanks! 

Allso, can confirm the zoom level is saved after clicking that button.

thelastWallE, 2021-05-05:
hmm i think you are actually on the version from yesterday. i had this letter in this version.
have you restarted after installing?

jalabaya, 2021-05-05:
Yes, I took your most recent link `https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/heads/rc.zip `and installed that, after that I was prompted to restart, which I did. I just did the procedure again (from the most recent link) and noticed in the log:
```
Found existing installation: OctoKlipper 0.3.8rc8
Uninstalling OctoKlipper-0.3.8rc8:
Successfully uninstalled OctoKlipper-0.3.8rc8
Successfully installed OctoKlipper-0.3.8rc8
Done!
```

So, I guess it remained the same?

thelastWallE, 2021-05-05:
It's just reinstalling, because the version is the same. But the latest commits are in the zip.

jalabaya, 2021-05-05:
Ok, well, after the restart (from your most recent link) it still shows the white "s".

thelastWallE, 2021-05-05:
There is a new function in octoprint for the logs to download.
Should be at http://"YOUR OCTOPRINT IP"/downloads/systeminfo.zip or over the system informaton dialog.
The link for the button should be in the bottom right of octoprint.
I suspect a cached file that is not updated correctly.
Maybe try to CTRL+F5 the site in the browser to get the latest files from octoprint.

jalabaya, 2021-05-05:
CTRL+F5 fixed it. Thank you for the help!

Sineos, 2021-05-05:
Two more (latest RC branch) 😏 

[image]

[image]

thelastWallE, 2021-05-05:
i need a bigger snip of the last image. you can also post the cfg here. So that i can investigate.
On my cfg is it all looking fine. 🧐

Sineos, 2021-05-05:
<details><summary>Complete cfg</summary>

```
[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
microsteps: 64
full_steps_per_rotation: 400
#rotation_distance: 40
rotation_distance: 39.95
endstop_pin: P1.29
position_endstop: 264
position_max: 264
homing_speed: 30

[tmc2209 stepper_x]
uart_pin: P1.10
interpolate: true
sense_resistor: 0.110
run_current: .955
hold_current: .5
stealthchop_threshold: 1000
#driver_TBL: 3
#driver_TOFF: 5
#driver_HSTRT: 2
#driver_HEND: 3

[stepper_y]
step_pin: P0.19
#dir_pin: P0.20
dir_pin: !P0.20
enable_pin: !P2.8
#step_distance: .0015589154812192
#step_distance: .0015625
microsteps: 64
full_steps_per_rotation: 400
#rotation_distance: 40
rotation_distance: 40
endstop_pin: P1.28
position_endstop: 250
position_max: 250
#position_min: -11.2
homing_speed: 30

#[tmc2209 stepper_y]
#uart_pin: P1.9
#interpolate: true
#sense_resistor: 0.110
#run_current: 1.00
#hold_current: 0.500
#stealthchop_threshold: 1000
#driver_TBL: 3
#driver_TOFF: 5
#driver_HSTRT: 1
#driver_HEND: 3

[tmc5160 stepper_y]
cs_pin: P1.9
spi_software_miso_pin: P0.5
spi_software_mosi_pin: P1.17
spi_software_sclk_pin: P0.4
interpolate: true
sense_resistor: 0.110
run_current: 1.00
hold_current: 0.500
stealthchop_threshold: 9999
#driver_TBL: 3
#driver_TOFF: 5
#driver_HSTRT: 1
[... 427 lines cut ...]
#*#
#*# [bed_mesh 100deg]
#*# version = 1
#*# points =
#*# 	  -0.072964, -0.060688, -0.059194, -0.047593, -0.040840, -0.041060, -0.047271, -0.061040, -0.092314
#*# 	  -0.085913, -0.058345, -0.070269, -0.038511, -0.039551, -0.047095, -0.073491, -0.082324, -0.122139
#*# 	  -0.086660, -0.076729, -0.064336, -0.048750, -0.037793, -0.047827, -0.058184, -0.074429, -0.103799
#*# 	  -0.083599, -0.068013, -0.059092, -0.043125, -0.043374, -0.060352, -0.058975, -0.073403, -0.107036
#*# 	  -0.073535, -0.063003, -0.054390, -0.030425, -0.049907, -0.057349, -0.043052, -0.062651, -0.095259
#*# 	  -0.080596, -0.060396, -0.046963, -0.028784, -0.029121, -0.035669, -0.041265, -0.053013, -0.085576
#*# 	  -0.074897, -0.067866, -0.060513, -0.041030, -0.039170, -0.027729, -0.032827, -0.055254, -0.071162
#*# 	  -0.052295, -0.032783, -0.034028, -0.022617, -0.014751, -0.026968, -0.020874, -0.025811, -0.063662
#*# 	  -0.034644, -0.007280, -0.006357, 0.027891, 0.027993, 0.012363, 0.009214, -0.018662, -0.044370
#*# tension = 0.2
#*# min_x = 0.0
#*# algo = bicubic
#*# y_count = 9
#*# mesh_y_pps = 2
#*# min_y = 11.2
#*# x_count = 9
#*# max_y = 245.36
#*# mesh_x_pps = 2
#*# max_x = 218.56
```
</details>

thelastWallE, 2021-05-06:
Ok can you try rc9? Should fix all left over problems.

closed 2021-05-07 by thelastWallE

Sineos, 2021-05-08:
Works nicely. Many thanks 👍

jalabaya, 2021-05-25:
Hello, found one more: 
[image]

thelastWallE, 2021-05-28:
Sry for the late reply. I didn't see the issue because it is closed.
I will add those gpio pins to the list. Seems like there is also more to add, because some are having hosts with other gpiochip numbers. ref:https://github.com/KevinOConnor/klipper/pull/2935#

commit 22fb1dd4 (2021-04-25): more highlight work
-closes #48