thelastWallE/OctoprintKlipperPlugin #50: [Feature Request] Option to move NavBar messages to Sidebar

charredchar, 2021-05-02:
First I must thank everyone working on this plugin, I saw the new update a month ago and, though it seems slightly on the surface, has been really great for day to day use!

But one thing that has really been bugging me with OctoKlipper... The Status messages that popup in the NavBar. I found them useful to have as I am in other tabs in OctoPrint but some times it is just such a long string that ends up in the NavBar and it completely destroys the layout at times. Would it be possible to have a secondary option for the status to either have it in the NavBar or to move it to the Sidebar where the Macros and such are? I feel this would help keep things a lot cleaner.

thelastWallE, 2021-05-03:
You can test it with https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.8rc6
I also truncate the messages that are over 36 chars for the navbar now.

charredchar, 2021-05-04:
Just installed rc6, looking nice! Sadly the config screen doesn't show anything anymore.

[image]

thelastWallE, 2021-05-04:
Were you on 0.3.8rc5 before?
I install now UI customizer. Because i can not find any bug that would cause this.

thelastWallE, 2021-05-04:
yeah ok. seems like there is some incompatibility with Plugin UIcustomizer.
I will look into that.

thelastWallE, 2021-05-04:
Can you test https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.8rc7 ? This should work for UIcustomizer.
I also changed the linkcolor on the sidebar.
I saw that you mentioned it on discord.

thelastWallE, 2021-05-12:
closed with version0.3.8

closed 2021-05-12 by thelastWallE