thelastWallE/OctoprintKlipperPlugin #52: new release not work, blank page

luco85, 2021-05-08:
hi! i installed new update but now not work. When i open octoklipper i have a blank page. After a system reboot octoprint was in safe mode.  I tried to uninstall and reinstall but nothing change. 
Octoprint 1.6.0
Octopi 0.18
Reload last change and reload from file if pressed nothing appears

[image]

thelastWallE, 2021-05-08:
Try to CTRL+F5 to get the uncached version from OctoPrint.
What browser are you using?
Can you me the octoprint.log or plugin_klipper_debug.log

luco85, 2021-05-08:
safari 14.1, tried to reset cache but persist
with chrome it work!
before this release no issue  (only not saving after modify .cfg )

[octoprint.log](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/6445976/octoprint.log)
[plugin_klipper_debug.log](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/6445977/plugin_klipper_debug.log)

thelastWallE, 2021-05-08:
Can you try this version with safari? 
Install with pluginmanager->get more->url option
https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/heads/devel.zip

luco85, 2021-05-08:
same issue

thelastWallE, 2021-05-08:
i will look into it again in 2hours. i think i discovered the problem but i am not at the computer atm.

luco85, 2021-05-08:
No prob! Thanks for your time!

thelastWallE, 2021-05-08:
Ok you can try https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/heads/testSafari.zip
Tomorrow i can test this with browserstack myself.

luco85, 2021-05-09:
the issue persist in safari. with chrome it work

wildchron, 2021-05-10:
Same issue here. Safari 14.0.3 on MacOS 11.2.3

thelastWallE, 2021-05-11:
**I like that you try this version before i release this.**
https://github.com/thelastWallE/OctoprintKlipperPlugin/archive/refs/heads/master.zip

I tested it on safari, firefox and chrome. I think i fixed it.
In safari the whole settings dialog of octoprint looks odd.

luco85, 2021-05-12:
Yep! It seems to be ok!

wildchron, 2021-05-12:
Working for me! I should mention that I use UI Customizer plugin also.

closed 2021-05-12 by thelastWallE