thelastWallE/OctoprintKlipperPlugin #53: Backing up the printer.cfg within Octoprint-Backup?

nibblerrick, 2021-05-10:
Hi,

thanks very much for taking care of this plugin.
I wonder if it could be possible to backup the klipper-printer-configuration when an octoprint-backup is being made.
Almost all important stuff for restoring to a working system is made with restoring an octoprint-backup (at least for me), only the klipper-printer-configuration is missing in this.
If I overlooked something in the other issues why this won't happen or if there are good reasons why this is a bad idea feel free to close this issue right away. Otherwise when this is a valid request I would be glad if you consider it in the future.

thelastWallE, 2021-05-10:
I think i will integrate a failsave feature for this. If someone saves the actual config as a failsave it will be written to config.yaml as a setting.

And then after a restore you can just load the failsave out of the config.yaml with a button.

thelastWallE, 2021-05-10:
hmm i think i will save another setting for changes of printer.cfg
For now i do this only if the parsing check isn't ok.

closed 2021-06-25 by github-actions[bot]