thelastWallE/OctoprintKlipperPlugin #54: No display Klipper Configuration

petitxef, 2021-05-12:
Hi!

First of all said I'm using Octoklipper 0.3.8.1, Octoprint 1.6.1, Octopi 0.18 on firefox 88.0.1

After update 0.3.8.1 the klipper configuration doesn't shows:

[image]

The others tabs, all correct.

thelastWallE, 2021-05-12:
Please try version 0.3.8.2
I did tested the new version with firefox yesterday but somehow i had a cached version.
Thanks for your patience.

petitxef, 2021-05-12:
Thanks!

It's works now (firefox & Chrome)
[image]

Thanks

closed 2021-05-12 by petitxef

thelastWallE, 2021-05-12:
Is this UIcustomizer with the cyborg theme?
I don't have a scrollbar on the far right. the whole button area on top should stay there if you scroll down.

petitxef, 2021-05-12:
yes, I've UIcustomizer with the cyborg theme

thelastWallE, 2021-05-13:
i discovered the problem with the scrollbar.
I had for me in the settings of UIcustomizer following css:
```
body.UICResponsiveMode #settings_dialog_content {
	height: calc(100% - 35px);
	margin-right: -18px;
	margin-top: 50px;
	width: calc(100% - 15px);
}
```
This fixed the height of the settings tab.
I fix it with the next version.

petitxef, 2021-05-14:
good! :)