thelastWallE/OctoprintKlipperPlugin #60: Latest Update completely wipes the config

VoltexRB, 2021-05-26:
I dont know what you changed, but after updating my config file is straight up gone. Not gone as in not recognized by Klipper anymore, but gone as in not even in the files any longer...
Reload last Changes and Reload from File also does nothing
   
[image]

thelastWallE, 2021-05-27:
On which version there you before?

There also should be backups on your config path. eg. `printer-date_time.cfg`

VoltexRB, 2021-05-27:
Whatever the latest version was before that, I've always kept it up to date.
I also recovered it from a backup, but the original file was gone. This is more of a heads up than an unsolved issue

thelastWallE, 2021-05-27:
I don't see a way that it would save like this from my side. I was also searching how it maybe would end up like that.
The only possible way would be that it got corrupted on saving on the sd card. There is also an unidentified symbol on the far right in your picture. So i think it got corrupted.

thelastWallE, 2021-05-27:
Nothing in the log maybe?

VoltexRB, 2021-05-27:
Thats the scroll bar compacted with both arrows overlapping.
Nothing in the logs either.
Also since this is fixed on my side and seems to probably have been coincidentally at the same time im closing this.
If no one else mentions this it probably was, if they do, nothing wrong with just opening another issue

closed 2021-05-27 by VoltexRB

thelastWallE, 2021-05-27:
Changes between 0.3.8.2 and 0.3.8.3 were: https://github.com/thelastWallE/OctoprintKlipperPlugin/compare/thelastWallE:0.3.8.2...0.3.8.3