thelastWallE/OctoprintKlipperPlugin #62: Since plugin install - Strange behaviour in file view

pandel, 2021-08-30:
Hi!

OctoPrint 1.6.1
Python 3.7.3
OctoPi 0.18.0
OctoKlipper 0.3.8.3

Since I installed the plugin, I get strange effects in any file view. I attach a screenshot, as it is a little hard to explain.

[image]

Regards,
Holger

thelastWallE, 2021-08-31:
Hallo, brauche noch mehr Infos. 
Liste von installierten Plugins. 
Welcher Browser?

Problem eingrenzen/beheben:
1. Webseite komplett neuladen mit Strg+F5
2. Plugin temporär abschalten über den kleinen Schalter in der Pluginliste in der Pluginverwaltung.

Ich wüsste nicht wie Octoklipper diese Listen beeinflussen könnte, aber man weiß ja nie.

pandel, 2021-08-31:
Hi!

Danke für die Denkanstöße :-) - daran, das Plugin zu deaktivieren, habe ich nicht gedacht. Das mache ich so gut wie nie. Es ist wohl in der Tat so, dass sich Octoprint so merkwürdig verhält, auch ohne dein Plugin.

Tut mir sehr leid, dich damit belästigt zu haben. Ansonsten ist dein Plugin nämlich wirklich super!

Bleib gesund, ich mache hier zu.
Gruß
Holger

closed 2021-08-31 by pandel

pandel, 2021-09-01:
Hi!

Nur abschließend zur Info, ich habe das Problem gefunden: wenn in Octoprint der SD Karten Support eingeschaltet ist UND in der Klipper Konfiguration 'virtual_sdcard' eingeschaltet ist, dann führt das zu diesem unkontrollierten Unfug. Vielleicht wäre es gut, wenn OctoKlipper bei der Installation darauf hinweisen würde :-)

Ich habe den SD Karten Support in Octoprint einfach abgeschaltet, die Klipper Konfiguration aber so belassen, wie sie ist. Damit ist alles gut bis dato.

Lieber Gruß
Holger

thelastWallE, 2021-09-01:
Danke für die Info! Wäre theoretisch möglich ne Abfrage zu machen dafür beim speichern der Config.