thelastWallE/OctoprintKlipperPlugin #66: By default, any config errors result in whole config not being saved (frustrating!!!!)

TTN-, 2021-10-01:
As the title says, by default, any single parsing error will cause the whole config to dispose of any changes.
I just spent the last hour meticioulsly going over all my config, I go to save it and the whole thing is cleared.

[image]

I just wanted to save to make sure I didn't loose progress as I'm working ... and well, now its gone.

Solution: can we please at least save the file, but still throw the error? or give an option?

TTN-, 2021-10-01:
Ok reload last changes restores them. Phew!!!

closed 2021-10-01 by TTN-

thelastWallE, 2021-10-02:
If you want to save without to check for parsing disable `Check parsing on save`.
The checkbox for this is where the editor dialog is.

I also changed the whole behavior in the new version. 
The editor is in an extra dialog there so it will be not closed automatically.