thelastWallE/OctoprintKlipperPlugin #67: Setting in config.yaml are lost!

axl12ap, 2021-11-02:
Hello,
just installed the last version of Octoklipper, the settings configured are lost event after press buton SAVE in octoprint settings.
Try to change any settings, then press SAVE button, reboot system in raspeberry, but the file config.yaml doesn't contain any information under plugin section of "klipper", only "_config_version: 3" present.

Octoprint: 1.7.1 (even older 1.7.0)
Plugin version: 0.3.8.3

Any help please?

thelastWallE, 2021-11-03:
Hi,
did you have OctoKlipper installed before or is this a fresh install of 0.3.8.3?

axl12ap, 2021-11-03:
> Hi, did you have OctoKlipper installed before or is this a fresh install of 0.3.8.3?

fresh install of entire RPI S.O. with octoprint, then installed klipper and octoprint plugins (I think you can see listed inside the yaml file)... also your latest version plugin..

axl12ap, 2021-11-03:
@thelastWallE if I want to try edit config.yaml and insert manual configuration plugins params... what are the params should I need to add? names and values..

thelastWallE, 2021-11-03:
like this

```
configuration:
      debug_logging: true
      fontsize: 20
```
you need to use spaces as indentation
Full list would be here: https://github.com/thelastWallE/OctoprintKlipperPlugin/blob/e4582265806b6b9679442f8818baad6f750e7117/octoprint_klipper/__init__.py#L104-L136

I would need to have the system info bundle to get to the problem of this.
It's in the right bottom corner. You can download a zip there.
I would like to know which python version.

axl12ap, 2021-11-03:
> I would need to have the system info bundle to get to the problem of this. It's in the right bottom corner. You can download a zip there. I would like to know which python version.

Attacched the ZIP you requested, thanks for your support.

env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true

Now I try the "manual" configuration..

axl12ap, 2021-11-03:
I've a problem after change param values, when I restart octoprint server it says in LOG:
2021-11-03 11:49:57,660 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart

> 2021-11-03 11:49:57,831 - octoprint.server - INFO - Shutting down...
> 2021-11-03 11:49:57,839 - octoprint.server.api.system - WARNING - Command for core:restart failed with return code -15:
> ! STDOUT: 
> ! STDERR: 
> 2021-11-03 11:49:58,175 - octoprint.server - INFO - Calling on_shutdown on plugins
> 2021-11-03 11:49:58,177 - octoprint.events - INFO - Processing shutdown event, this will be our last event
> 2021-11-03 11:49:58,181 - octoprint.events - INFO - Event loop shut down
> 2021-11-03 11:49:58,199 - octoprint.server - INFO - Goodbye!

Attacched my changed config.yaml, can you check please?

[manual_config.zip](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/7467040/manual_config.zip)

thelastWallE, 2021-11-03:
Better to delete the lines for octoklipper and restart OctoPrint. Unfortunately i don't have time at the moment. Seems like it has a problem with a symbol in the configuration file. I need to look into that in the evening.

axl12ap, 2021-11-03:
> Better to delete the lines for octoklipper and restart OctoPrint. Unfortunately i don't have time at the moment. Seems like it has a problem with a symbol in the configuration file. I need to look into that in the evening.

Thanks so much for your time.. let me know any news, or the correct params values to manually insert into config

thelastWallE, 2021-11-03:
Seems like the problem is with the printer.cfg. Can you share it? Looks like the encoding is different from what to expected.

thelastWallE, 2021-11-03:
You can also try to install the upcoming version https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.9rc5
Install the attached zip with the Plugin Manger in OctoPrint.

The saving and loading of the config there is separate from the plugin settings saving.

axl12ap, 2021-11-03:
> Seems like the problem is with the printer.cfg. Can you share it? Looks like the encoding is different from what to expected.

That's exactly the problem.. I found the same issue, if I delete my files your plugin works "correctly" (even doesn't load my configuration).

If you found the issue, let me know please.. I'll investigate

[Printer.zip](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/7468254/Printer.zip)

axl12ap, 2021-11-03:
> You can also try to install the upcoming version https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.9rc5 Install the attached zip with the Plugin Manger in OctoPrint.
> 
> The saving and loading of the config there is separate from the plugin settings saving.

Does it work right now?? or still in development (and bugfixing)?

thelastWallE, 2021-11-03:
Ok, found the problem. Your file seems to be encoded with latin-1, ISO-8859-1. And cannot be loaded as utf-8.
I will try to load it as latin-1 after it could not be loaded with utf-8.
Will push a hotfix to master.

balkin, 2021-11-04:
Actually might be not the best solution. What if the file is encoded in ISO-8859-5, for example? Or in Korean encoding (I'm pretty sure they have alphabet also). Maybe the encoding can be determined from the system locale, or elsewhere?

axl12ap, 2021-11-04:
> Ok, found the problem. Your file seems to be encoded with latin-1, ISO-8859-1. And cannot be loaded as utf-8. I will try to load it as latin-1 after it could not be loaded with utf-8. Will push a hotfix to master.

Ok, thank FYI ...I normally use Notepad++ for editing source code files, and also txt files included klipper configurations, but I never noticed how the file was encoded when saving. If required I can save all .cfg files in UTF-8 format.

> Actually might be not the best solution. What if the file is encoded in ISO-8859-5, for example? Or in Korean encoding (I'm pretty sure they have alphabet also). Although Germany, Italy and some other Western European countries share ISO-8859-1, there's for example Czechia and Poland with ISO 8859-2. Maybe the encoding can be determined from the system locale, or elsewhere?

I agree with what you wrote, but I do not understand how the klipper firmware can instead correctly interpret each encoding of the configuration CFG file, always assuming that it reads all the encodings

axl12ap, 2021-11-04:
> > > You can also try to install the upcoming version https://github.com/thelastWallE/OctoprintKlipperPlugin/releases/tag/0.3.9rc5 Install the attached zip with the Plugin Manger in OctoPrint.
> > > The saving and loading of the config there is separate from the plugin settings saving.
> > 
> > 
> > Does it work right now?? or still in development (and bugfixing)?
> 
> Should be pretty stable, but it will not fix your problem completly.
> 
> Ok, found the problem. Your file seems to be encoded with latin-1, ISO-8859-1. And cannot be loaded as utf-8. I will try to load it as latin-1 after it could not be loaded with utf-8. Will push a hotfix to master.

guysoft, 2021-11-04:
Hey, there is a release saying this is fixed but its reopened - does the new release fix the issue?

thelastWallE, 2021-11-04:
If your file is encoded with ISO 8859-1 then yes.

I will work on a better solution to handle other encodings, too.

closed 2021-11-04 by thelastWallE