thelastWallE/OctoprintKlipperPlugin #72: Repeating errors cover screen and force reload.

charredchar, 2021-11-17:
Running OctoPi 0.18.0, OctoPrint 1.7.2 and OctoKlipper 0.3.9.
The new "feature", Popups for Errors from Klipper, floods the screen and forces the need of disconnecting OctoPrint from Klipper and a reload of the page to clear it. This particular popup is caused by my printer being turned off and Klipper stating it can not connect to it which is a very common state.
At a minimum there needs to be an option to turn off popups from OctoKlipper, preferably not have it on by default.

[image]

thelastWallE, 2021-11-17:
Will push a fix for it today.

charredchar, 2021-11-18:
Thank you for adding the option! I also see that you introduced the modalish flow so even with it on by default you don't get flooded and it gives a chance to get into the settings to disable notifications.

This is only slightly related so I can create a new issue if you prefer.

closed 2021-11-18 by charredchar

thelastWallE, 2021-11-20:
will add a max height setting for it.