thelastWallE/OctoprintKlipperPlugin #79: Save_config command not work after last update

luco85, 2022-01-16:
B
[klippy.log.zip](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/7877678/klippy.log.zip)
efore submitting your issue please make sure to provide the following information

Octoklipper v.0.3.9.4
Octoprint v. 1.7.2

klipper last version

Seems that after issue a save_config it not save data to config file. I see that mine bed_mesh not changes after run bed mesh calibrate. after some try i deleted the automatic data generated in printer file and restart bed mesh. after probing my probe macro issue save_config and after klipper firmware restart there wasn't bed mesh data in config file

luco85, 2022-01-16:
> [klippy.log.zip](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/7877678/klippy.log.zip) efore submitting your issue please make sure to provide the following information
>
> Octoklipper [...]

luco85, 2022-01-16:
I found that the data of bed mesh are writed on the backup printer file and no in the printer.cfg. This is a normal behavior
?

thelastWallE, 2022-01-17:
Hi,
it should detect a save_config command and asks if you want refresh the editor after it. 
Let me check if i missed some behavior that prevents the command to be caught.

luco85, 2022-01-18:
Hi! The save_config is issue automatically at the end of macro g29. No issue till this version.

thelastWallE, 2022-01-18:
So the gcode command would be "`G29`" that got send from OctoPrint. And Klipper then internally runs this macro and executes SAVE_CONFIG at the end. The thing is that OctoPrint is not knowing that there was a `SAVE_CONFIG` command.
I checked again in Klipper and as i just remembered correctly is that Klipper is not giving a response over the terminal after `SAVE_CONFIG` and just does a restart. So i can't easily catch the command and know that the cfg was changed to reload it.

~~After a little playing this through in the head the best i can do is catch the "`G29`" and then the connection gets lost i know that Klipper is restarting and then ask if one wants to reload the cfg.~~

~~I think i will create a PR for a gcode response just before klipper restarts or after the SAVE_CONFIG has written the new cfg.~~

-> We can actually use this: `RESPOND TYPE=command MSG="SAVE_CONFIG"` from [Klipper G-Codes.md](https://github.com/Klipper3d/klipper/blob/master/docs/G-Codes.md#send-message-respond-to-host). Insert the `[respond]` section in the cfg to enable this.
This would be placed in the G29 macro just before the `SAVE_CONFIG` command.
I can check then for a response from Klipper that is: `// SAVE_CONFIG` and pop a dialog that is requesting a reload of the cfg file.
I will work on it in the next 2 days.

luco85, 2022-01-19:
Ok! Thanks for your time

commit 850447fd (2022-01-20): :sparkles: add additional check for SAVE_CONFIG

- adds an additional listener for "// SAVE_CONFIG" responses from
printer to pop a dialog which informs over the external chnged config.

fixes #79
closed 2026-09-12 by thelastWallE