thelastWallE/OctoprintKlipperPlugin #80: after termination Unknown command:"M108

mczeus, 2022-02-02:
I switched from Marlin to klipper, but have now reinstalled everything to be fresh.
I've set everything and there are no error messages either, except when I cancel the print then I get an "Unknown command:"M108"
although I turned it off in the settings.
I slice with Simplify3D and took out all start and end Gcode there, but always get an Unknown command:"M108"
When the print goes through and is finished I can't drive an axis because I have to do homeing first, why is that?

thelastWallE, 2022-02-02:
You can remove the M108 gcode in the OctoPrint Settings. It's under Printer-> Serial Connection -> Firmware & protocol -> Protocol fine tuning -> Emergency commands.

mczeus, 2022-02-02:
I have done there is only "M112, M410"

closed 2022-03-10 by github-actions[bot]