thelastWallE/OctoprintKlipperPlugin #81: Moonraker Compatibility - Add configurable path for printer.cfg and klippy.log

jalanjarosz, 2022-02-04:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I noticed in the init.py file that the source path to both printer.cfg and klippy.log are  hard coded:

```
            configuration=dict(
                debug_logging=False,
                config_path="~/",
                baseconfig="printer.cfg",
                logpath="/tmp/klippy.log",
```

In order to support Moonraker (Klipper Web API), both the printer.cfg and klippy.log file are moved to the /home/pi user directory.  Moonraker suggests: ~/printer_config and ~/printer_logs while the Mainsail WebUI suggests ~/klipper_config and ~/klipper_logs.

I'd suggest making the path to these files user configurable.

**Describe alternatives you've considered**
None

**Additional context**
The OctoprinterKlipperPlugin could use the Moonraker API for additional enhancements, but this change would be the first step needed followed by installing Moonraker.

For reference: https://github.com/Arksine/moonraker

thelastWallE, 2022-02-04:
The config path is configurable through the settings dialog of OctoPrint.

thelastWallE, 2022-02-04:
I will add a second setting for the klippy logfile.

thelastWallE, 2026-09-03:
Checked again. The logfile path is already in the current version 0.3.9.5

closed 2026-09-03 by thelastWallE