thelastWallE/OctoprintKlipperPlugin #90: Always have to firmware restart, and reboot printer/octoprint after each big print.

DuckersMcQuack, 2022-10-04:
Before submitting your issue please make sure to provide the following information

About your environment:

    Octoprint version 1.8.4

    An OctoPrint log file. Included
    
Thanks
[octoprint.log](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/9707637/octoprint.log)
[plugin_klipper_debug.log](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/9707638/plugin_klipper_debug.log)

Continuous issues whenever long print is done, same when attemting to connect to the printer and i ALWAYS have to reboot the printer and restart octoprint, then firmware_restart command.

DuckersMcQuack, 2022-10-18:
OctoKlipper
Printer Profile
Klipper state: Shutdown
Macros
State
State: Connecting
File:  
Uploaded:
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: -
Printed: -
Files
Zelda Sword full size
Size: 20.5MB
30 mm cube_6m_0.20mm_205C_PLA_ENDER3PRO.gcode
Uploaded: a minute ago
Size: 37.8KB
>
ironhead2.gcode
Uploaded: 2 hours ago
Size: 3.8MB
>
ironhead.gcode
Uploaded: 2 hours ago
Size: 4.2MB
>
Top_Joy_Fidget_Low_Tolerance_21m_0.20mm_205C_PLA_ENDER3PRO.gcode
Uploaded: 3 hours ago
Size: 643.1KB
>
Calibration Cube_43m_0.20mm_205C_PLA_ENDER3PRO.gcode
Uploaded: 4 hours ago
Size: 684.6KB
>
CFFFP_Calibration Cube.gcode
Uploaded: 11 hours ago
Size: 719.6KB
>
CFFFP_Bed_Leveling_five_squares.gcode
Uploaded: a day ago
Size: 719.5KB
>
CFFFP_Iron Man Snap Torso.gcode
Uploaded: a day ago
Size: 29.9MB
>
CFFFP_Iron Man Snap base.gcode
Uploaded: 2 days ago
Size: 77.6MB
>
CFFFP_Iron Man Snap Right Arm.gcode
Uploaded: 3 days ago
Size: 27.5MB
>
CFFFP_Iron Man Snap Left Arm.gcode
Uploaded: 3 days ago
Size: 21.3MB
>
CFFFP_Iron Man Snap Legs.gcode
Uploaded: 4 days ago
Size: 52.5MB
>
CFFFP_duck case6.gcode
Uploaded: 6 days ago
Size: 2.9MB
>
CFFFP_FilamentClipV2.gcode
Uploaded: 6 days ago
[... 871 lines cut ...]
01:53:30
Too many consecutive timeouts, printer still connected and alive?
01:53:30
Klipper: Disconnected from host
01:54:01
Klipper: Standby
02:05:36
Restarting Klipper.
02:05:44
Klipper: Connecting ...
 
Restart
Tools
Macros

    OctoPrint 1.8.2 Python 3.7.3 OctoPi 0.18.0

   [ Web](https://octoprint.org/)[ Forum](https://community.octoprint.org/)[ Discord](https://discord.octoprint.org/)[ FAQ](https://faq.octoprint.org/)[ Docs](https://docs.octoprint.org/)[ About](javascript:void(0))[ System](javascript:void(0)) 

Communication error
There was a communication error while talking to your printer. Please consult the terminal output and octoprint.log for details. Error: Too many consecutive timeouts, printer still connected and alive?

____________

This error is what i get now.

vordenken, 2023-05-24:
I think I have the same problem? 
Every time I start the printer, OctoKlipper tries to connect but I can see somethings wrong because I get reported 0degrees celsius for the hotend and bed. After restarting the firmware using the button in OctoKlipper, it starts working.
I have to do this every time I turn the printer off/on. 
I don't have these issues on moon raker.