thelastWallE/OctoprintKlipperPlugin #92: [Feature Request] Control the color of custom macro buttons

robertfchapman, 2022-12-16:
Currently, all custom macro buttons are the same color.  It would be nice if you could define the color of the macro button when creating it.

Mabe, add a color input field that has a default color setting in it that can be changed.  This would allow one to change, say the E-Stop button to "#FF0000" (RED)

This would also help organize/group macro buttons by color.

SinisterRj, 2023-02-22:
Also create colapsible user defined classes will be nice. So users can have more macro buttons in the same space. Like "Tests", "Calibration", etc

thelastWallE, 2023-02-22:
Sry guys. E-Mail notification is way off, so I just have read about it. Looks relatively easy to do.

closed 2026-09-03 by thelastWallE