thelastWallE/OctoprintKlipperPlugin #94: Connect button does not do anything.

cray12399, 2023-02-25:
**Distribution version. Eg. octopi, armbian**
Raspian 64-bit lite with Octoprint installation.

**Octoprint version**
1.8.6

**An OctoPrint log file. You can access and download it in the OctoPrint settings under the "Logging" menu entry.**

[octoprint.log](https://github.com/thelastWallE/OctoprintKlipperPlugin/files/10830007/octoprint.log)

**The issue:**
Hi I recent installed Octoprint and Klipper on a Raspberry Pi 3 and SKR Mini E3 v3.0 using the following as guides:
https://www.klipper3d.org/Installation.html
https://www.youtube.com/watch?v=0mOjDtQzrTk

Despite follow all of the steps perfectly to the best of my knowledge, the connect button in Octoprint does nothing when OctoKlipper is enabled. I am able to click it, but I don't get a popup, an error, a terminal log, nothing when I click it. For all intents and purposes, it is a dead button. From what I can tell, even if there are connection issues, the button should at least give me an error, right? Throughout all of this, Klipper is perpetually on "Standby."

Additionally, it is worth nothing that my connection appears to be somewhat fine, as I have an Octoprint menu on my Ender3 display. 

What steps should I take in diagnosing this issue, and what kind of additional information can I provide to help you best support me in this issue?

Thank you.

cray12399, 2023-02-25:
Update: I am getting the following error when I inspect the page and click the connecct button:

`POST http://172.20.2.242:5000/api/connection [HTTP/1.1 400 BAD REQUEST 47ms]`

dekahobby, 2023-02-27:
same with me, but when i check "Auto-connect to printer on server start" and restart VM ( im using ubuntu server with mini PC) octoklipper can connect, so i need restart VM to connect

ps - i have mainsail too on the same machine

thelastWallE, 2023-02-27:
The  problem is your connection from klippy the host software of klipper to the firmware of klipper on your printer. Klippy is giving the error “!! mcu 'mcu': Unable to connect”.you need to setup your mcu name in the printer.cfg under [mcu].Frequently Asked Questions - Klipper documentationklipper3d.orgBest regards

https://www.klipper3d.org/Installation.html
https://www.youtube.com/watch?v=0mOjDtQzrTk

thelastWallE, 2023-02-27:
Did you use kiauh by chance? The printer.cfg that klippy is loading is at a different place if you used kiauh to install klipper.

cray12399, 2023-02-28:
@thelastWallE Hi, thank you for responding to my issue! 

So in the logs it says there is an issue with the serial port connection, but I got it the same way as the klipper docs specify:

`ls /dev/serial/by-id/*`

Also, I used Kiauh, but I edited both the configuration file in the Klipper config directory AND the one in the printer_data directory. As well as editing it in the OctoKlipper interface using the editor.

Finally, I think my configuration file is correct because I just installed Fluidd, and it connects to my printer like a charm. (Not planning on using Fluidd full time though, just as a test.)

Larbaco, 2023-04-12:
> same with me, but when i check "Auto-connect to printer on server start" and restart VM ( im using ubuntu server with mini PC) octoklipper can connect, so i need restart VM to connect
> 
> ps - i have mainsail too on the same machine

you need to use the serial port created by your instalation methode. 
in my case using mainsailOS i have mine on:
Serial Port:  "/home/pi/printer_data/comms/klippy.serial"
Config folder:  "/home/pi/printer_data/config"
Klipper Log File: "/home/pi/printer_data/logs/klippy.log"

Ryaja2, 2023-08-09:
I have the same issue, and the exact config file and printer path works on mainsail and with octoklipper disabled.

git2212, 2024-10-08:
Is there a resolution to this? ... I am encountering the same problem with OctoKlipper. Octoprint is able to connect itself when the "replace connection panel" is not checked in the OctoKlipper plugin config :(